georgewhewell/undervolt #187: ERROR:root:Failed to apply core: set -49.8046875, read 0.0

UNRA1DUser, 2024-03-16:
Hi,

if I try to undervolt -> undervolt --gpu -50 --core -50 --cache -50 --uncore -50 --analogio -50

I get this Error Message -> ERROR:root:Failed to apply core: set -49.8046875, read 0.0

```
undervolt --read

temperature target: -0 (100C)
core: 0.0 mV
gpu: 0.0 mV
cache: 0.0 mV
uncore: 0.0 mV
analogio: 0.0 mV
powerlimit: 250.0W (short: 0.00244140625s - enabled) / 125.0W (long: 224.0s - enabled)
turbo: enable
```

I started my Linux in UEFI mode. Is that maybe a problem? Do I need to start it in legacy mode?

UNRA1DUser, 2024-04-20:
Can someone help me?

notstarboard, 2024-04-28:
@UNRA1DUser See [here](https://github.com/georgewhewell/undervolt/issues/91) - it seems like a pretty common issue. If you have the option to revert to older firmware, that might help you out. Otherwise, I'm not sure how to fix this. I'm stuck on this too.

GermanAizek, 2024-10-09:
Same error

DaniyalAhmadSE, 2024-10-13:
Resetting BIOS to factory settings, and disabling secure boot fixed this for me.

DaniyalAhmadSE, 2024-10-13:
It appears to be a copy of [Issue#91](https://github.com/georgewhewell/undervolt/issues/91)