georgewhewell/undervolt #188: Power litmit, what is long,short and time window ?

ForteDexe, 2024-05-12:
Sorry I am new, I am not sure what it means.

WinWeirdo, 2025-05-19:
Powerlimit is the internal limit that blockes power being drawn over the limit. long and short are the names of the two, one is short as for short duration and the other one is long as for longer duration. The time window is the amount of the seconds from being activited to stop limiting that amount and returning to normal powerlimit or shorter one

mathew2214, 2025-10-21:
is there a way to disable power limits entirely? let the system run as hard as it needs to for however long it needs to?

mathew2214, 2025-10-31:
yeah, it doesnt work. i set it to something that ill never actually hit, and throttling occurs anyways