georgewhewell/undervolt #191: PermissionError: [Errno 1] Operation not permitted when sudo undervolt --read

danielelongo14, 2024-08-05:
Hello, i got this error with Ubuntu 22.03 when executing the command:
temperature target: -5 (95C)
Traceback (most recent call last):
  File "/usr/local/bin/undervolt", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.10/dist-packages/undervolt.py", line 457, in main
    voltage = read_offset(plane, msr)
  File "/usr/local/lib/python3.10/dist-packages/undervolt.py", line 209, in read_offset
    write_msr(value_to_write, msr.addr_voltage_offsets)
  File "/usr/local/lib/python3.10/dist-packages/undervolt.py", line 82, in write_msr
    os.write(f, pack('Q', val))
PermissionError: [Errno 1] Operation not permitted

p0ll3, 2024-09-10:
did you find a fix for this issue?

danielelongo14, 2024-09-10:
No I'm not using Ubuntu anymore at the moment

JosefSchoenberger, 2025-01-05:
This most likely comes from the Lockdown LSM, which prevents direct access on MSRs from userspace. Lockdown is automatically enabled with secure boot, which in turn is enabled by default on Ubuntu.