georgewhewell/undervolt #192: PEP 668 - Usage of venv or pipx for installation for Debian 12 or Ubuntu 24.04

dengdenglele, 2024-08-16:
As of Debian 12 or Ubuntu 24.04 LTS it is not recommended to install any pip package system-wide via "pip install name-of-package", see PEP 668.

I would like to ask, if there are limitations when "undervolt" is installed via pipx or inside a virtual environment? Does undervolting work at all, when installed via pipx or in venv?

If so, do I still need to use sudo for commands such as "undervolt --read"?

It would be great, if the documentation and installations steps gets updated in accordance with the new guidelines.

SemKl, 2025-01-03:
I'm facing the same issue. Did you find out any way to solve this, or have any update on your side?

georgewhewell, 2025-01-03:
you can install it in user environment. you still need to use sudo to call it