georgewhewell/undervolt #194: Turbo booster ondemand.

felipe241916, 2025-01-24:
Hello, everyone.

It worked very well on my Avell MOB a70, but I would like to go back to the default. After uninstalling the package and reverting the settings, I realized that the turbo mode no longer works on demand. It stays on or off. Can you help me?

`root@felipe-work:/home/xxxx# undervolt -r
temperature target: -5 (95C)
core: 0.0 mV
gpu: 0.0 mV
cache: 0.0 mV
uncore: 8000.0 mV
analogio: 0.0 mV
powerlimit: 120.0W (short: 0.00244140625s - enabled) / 120.0W (long: 96.0s - enabled)
turbo: disable

`

aleb2000, 2025-03-07:
Hey, for curiosity I read what the script does, and it seems that all it does is set the attribute `/sys/devices/system/cpu/intel_pstate/no_turbo` to 1 to disable turbo, which should have no bearing on the turbo being on demand or not.
Looking at the [linux documentation](https://www.kernel.org/doc/html/v4.12/admin-guide/pm/intel_pstate.html#global-attributes):

> If set (equal to 1), the driver is not allowed to set any turbo P-states (see [Turbo P-states Support](https://www.kernel.org/doc/html/v4.12/admin-guide/pm/intel_pstate.html#turbo-p-states-support)). If unset (equalt to 0, which is the default), turbo P-states can be set by the driver. [Note that intel_pstate does not support the general boost attribute (supported by some other scaling drivers) which is replaced by this one.]

Maybe check the value with `cat /sys/devices/system/cpu/intel_pstate/no_turbo`.

Also from my understanding any change the this utility does should be temporary, and on reboot it should return to default, are you sure you are not inadvertently running it on startup automatically?

Otherwise I would say the problem is probably somewhere else, maybe check your bios settings to verify that everything is as you want it there.

felipe241916, 2025-03-07:
Thanks for your reply. So, after restarting, the modified values return to the default values, I suppose?
but it's not working like that. The CPU is in turbo mode (laptop fans at maximum speed) and the CPU's base clock is around 3.0GHz. Even with all the packages uninstalled...

I upgraded to ubuntu 24 LTS and the problem disappeared, but I'm not sure which file or files are responsible for this problem.

BTW... everything is fine now :).

Thanks for your help!

closed 2025-03-07 by felipe241916

aleb2000, 2025-03-09:
I imagined you might have already solved it since it was a month ago, but I saw your issue unanswered and I wanted to drop a message anyway, on the odd chance I might be able to actually help someone. I'm not that knowledgeable on how this tool works, but I've skimmed the code and I didn't see anything that from my understanding should persist across restarts, so I'm a bit stumped on that.

Tbh, I haven't really tried it myself, I actually wanted to use it, but I have Secure Boot enabled and turns out undervolting cannot be done on linux when it is enabled. And I don't want to disable it.

Cheers, and hopefully your computer will only turbo in moderation from now on 😄.