georgewhewell/undervolt #196: Doesn't apply to igpu on i5-8350u

nthpond288371, 2025-03-18:
all other options work as they cause the computer to crash when set to ridiculous levels except the gpu. it shows when doing the --read command but has no effect.

on a thinkpad x280
kubuntu 24.04.2 LTS
i5-8350u
intel uhd graphics 620

nxnen8282, 2025-03-18:
Our system attempted to address this issue:

Modified the voltage offset application logic to ensure GPU voltage changes are properly synchronized with core/cache changes. For integrated Intel GPUs, the voltage plane is tied to the core voltage plane and requires special handling.

You can examine the suggested modifications here: https://github.com/nxnen8282/georgewhewell-undervolt-196/commit/44ffbcae48570a475f9e2e1a569363baade871c0.

> [!CAUTION]
> Advisory: This solution was generated using automated tools, and we strongly recommend reviewing the code thoroughly before implementation. The solution may be partial - please consider it as inspiration for your own implementation.

---

http://missionfixinternet.com

BreadlyUwU, 2025-11-14:
It doesn't work either with the iGPU from the Intel i5-10300H, the script will works fine with the CPU and Cache but will have no effect to the iGPU.

(Dell G3 15 3500 using Arch (Linux 6.17.7))